hackerspace-silesia/Smogomierz #88: Duży rozrzut odczytów z czujnika pms7003

szczemp, 2020-03-03:
Zmontowałem zestaw i wgrałem soft. Przez kilkanaście dni testowałem w pomieszczeniu i wyglądało w miarę ok. Wartości wahały się pomiędzy 2ug do 30ug dla PM1.0, 2ug do 45ug dla PM2.5 i 2ug do 60ug dla pm10.0. Dziś wystawiłem go na balkon i rozrzut jest straszny. Raz jest 12, 14, 18 odpowiednio dla PM1.0, PM2.5 i PM10.0, a za godzinę 250, 560 i 740 odpowiednio dla PM1.0, PM2.5 i PM10.0.
Przy odczycie co kilka minut skacze trochę mniej, ale też wartości są wysokie np teraz mam 150, 270 i 325 odpowiednio dla PM1.0, PM2.5 i PM10.0.
Wiem, że wartości mogą się wahać, bo lekko zawieje i już smog ucieka. Ale czy takie wysokie wartości są ok? Sprawdzałem na mapie airly okoliczne czujniki i tam też wartości są pomiędzy 50 a 200, ale nie zauważyłem 700.

bfaliszek, 2020-03-03:
U moich rodziców, gdy zawieje dymem z sąsiedniego domu, to jest nawet 1500ug/m3. 
Jeśli miernik Airly jest >500m od Ciebie to może mieć nawet kilkaset ug/m3 różnicy. Nie ma w tym nic dziwnego.

szczemp, 2020-03-04:
Dzięki za odpowiedź.
Mam jeszcze inne pytanie. Czytałem gdzieś w opisie smogomierza, że potrafi się sam skalibrować, ale potrzebuje odczyt wilgotności powietrza. Problem u mnie jest taki, że użyłem czujnika bme280, którego używałem w innym projekcie już ponad 5 lat i wydaje mi się, że zaczął wariować. Na czym polega ta kalibracja?

bfaliszek, 2020-03-04:
W ustawieniach możesz włączyć automatyczną kalibrację. Zbija ona trochę odczyty PM przy wysokiej wilgotności powietrza(>60%): 
https://github.com/hackerspace-silesia/Smogomierz/blob/master/Smogomierz.ino#L1610-L1616

closed 2021-11-16 by bfaliszek