hackerspace-silesia/Smogomierz #90: Dwa problemy: brakujące pomiary oraz MQTT

tnowak, 2020-04-04:
Hej. W końcu znalazłem czas aby złożyć zestaw DYI kupiony w Botlandzie.
Dzięki zamieszczonej tu instrukcji dałem radę mimo, że z "kabelkami" w życiu miałem mało do czynienia.
Po wgraniu firmware (tego też musiałem się nauczyć) i uruchomieniu urządzenia zidentyfikowałem dwa problemy, z którymi nie mogę sobie poradzić:

1. Nie wiem dlaczego, ale nie widzę pomiarów temperatury i wilgotności.
Konfigurację mam zrobioną tak:

[image]

A pomiary wyglądają tak:

[image]

API też zwraca tylko zanieczyszczenia:

[image]

To samo z MQTT.
Czy coś robię źle?
Wydaje mi się, że kabelki podłączyłem zgodnie z instrukcją:

[image]

Mój drugi problem dotyczy konfiguracji MQTT:

[image]

Czy jest jakiś powód, dla którego smogomierz publikuje w ścieżce zaczynającej się od slasha?
Zgodnie z dokumentacją MQTT slashe rozdzielają poziomy, więc tu jakby poziom pierwszy jest pusty. Być może jest to dopuszczalne technicznie, ale na pewno dziwne. Mam ok 100 urządzeń publikujących w MQTT i żaden ich temat nie zaczyna się od slasha, tylko od nazwy poziomu.

Widziałem na różnych screenach tutaj, że wcześniej tak nie było. To błąd, czy jakaś zmiana?
Jeśli zmiana, to czy jest jakaś szansa, aby dodać do oprogramowania opcję "do not start MQTT topic with a slash"?

Dzięki z góry za pomoc.

bfaliszek, 2020-04-04:
1) to albo masz źle podłączone BME280 albo sam moduł BME280 jest wadliwy. Dlatego nigdzie nie pokazuje danych pomiarowych.

2) tak sugerowano mi we wcześniejszym wątku o MQTT - https://github.com/hackerspace-silesia/Smogomierz/issues/54. Jak rozumiem, chodzi Ci o to, żeby było `smogly/sensor/PM10/4` a nie `/smogly/sensor/PM10/4`?

tnowak, 2020-04-04:
Dzięki za szybką odpowiedź.

Ad.1. 
BME280 które dostałem ma 6 pinów, u Was na zdjęciach 4 - nie wiem czy ma to znaczenie. W każdym razie mam je podłączone tak:

```
BME   | ESP8266/NodeMCU
======+================
VCC ----> 3V
GND ----> G
SCL ----> D4
SDA ----> D3 
```

Pozostałe dwa piny na BME280 (SCB i SDO) są wolne.
Chyba więc jest dobrze.
No to by wynikało, że czujnik jest uszkodzony :/

Ad.2.
Dokładnie tak.

Jest jeszcze jedna "dziwność" w tej implementacji MQTT, właściwie z tej samej bajki co slash na początku, bo dotyczy slasha na końcu.
Chodzi o to, że wartość podana na ostatnim poziomie nie ma żadnej nazwy.
Czyli przykładowo jest tak:
smogly/sensor/PM10/=5
A na podstawie dużej liczby innych urządzeń MQTT które mam powiedziałbym, że standardem jest coś takiego:
smogly/sensor/PM10/value=5
Czyli "monitorowany" przez klienta temat zaczyna oraz kończy się nazwą poziomu (a nie separatorem).
Obecny format da się odczytać, ale po prostu zwracam uwagę, że to dziwna konstrukcja.

tnowak, 2020-04-04:
Update: jednak miałem jakiś problem u siebie w systemie z odczytem aktualnego formatu (ze slashem na początku i na końcu).
Tymczasowo zrobiłem w Node-RED konwersję na "standard"

[image]

To oczywiście głupie, bo dane trafiają do MQTT dwa razy i na około, więc fajnie byłoby gdyby można było w firmware skonfigurować sobie odpowiednio 🙏

tnowak, 2020-04-09:
Błażej, no niestety to nie była wina wadliwego czujnika BME280.
"Wykosztowałem" się na dwa kolejne BME280 (dwa aby mieć pewność) i NA ŻADNYM Wasz soft nie działa.
A wszystko podłączone zgodnie z instrukcją - sprawdzałem 10 razy. Kabelki też na wszelki wypadek wymieniałem.

Dla już absolutnej pewności natrudziłem się i zainstalowałem na płytce ESPeasy wersja Dev (bo ma też obsługę PMS7003).
Wszystkie trzy czujniki BME280 bez problemu są widziane przez ESPeasy. Oto dowód:

[image]

Tak naprawdę chciałem już machnąć ręką na Wasz soft i zostać na ESPEasy, ale na nim obsługa PMS7003 jest jeszcze bardzo szczątkowa i nie ma np. dobrze zrobionych cyklicznych pomiarów.
Jest tylko pomiar ciągły, a w ten sposób mój sensor pożyje tylko rok.

Tak więc bardzo uprzejmie proszę o pilną weryfikację tego błędu, zobaczenie co może być nie tak
i możliwie szybkie naprawienie.

jakubcichocki, 2020-04-09:
> Tak więc bardzo uprzejmie proszę o pilną weryfikację tego błędu, zobaczenie co może być nie tak
> i możliwie szybkie naprawienie.

Jak roszczeniowo się zrobiło w stosunku do projektu open source, ojoj. 

Spróbuj podłączyć się pod smogomierz b

akotas, 2020-04-09:
Jeśli chodzi o BME280 to raczej nie błąd w kodzie, no chyba ze coś poszło nie tak w kompilacji. wczoraj wgrywałem 2.7.5 do takiego samego świeżego zestawu i śmiga. 

Jak kolega Jakub mówi USB debug i logi, nikt tego za Ciebie nie zrobi. Jak by co wrzuć tu dump logów to się zobaczy.

tnowak, 2020-04-09:
To wszystko jest naprawdę dziwne. Wróciłem dzisiaj do firmware'u smogly, aby wykonać te debugi po USB (choć jeszcze nie wiem jak, ale nauczyłbym się) i jestem totalnie zaskoczony dwoma zdarzeniami:

1. Po wgraniu firmware, nie mogłem znaleźć telefonem hostspotu smogomierz-xxxxx.
Ku mojemu szczeremu zaskoczeniu, smogly już był podłączony do mojej sieci Wifi.
Myślałem że taki firmware flash robi orkę na chipie i wrzuca się w formie dziwiczej.
Jak przejście przy przejściu z ESP Easy do smogly uchowało się hasło do mojego wifi? 😮

2. Drugie zaskoczenie - zgadnijcie co?! Tak, dobrze się domyślacie (- TVP Style 😁)

[image]

Najciekawsze jest to, że wszystko jest IDENTYCZNIE podłączone. Specjalnie robiłem zdjęcia
wtedy i teraz, więc mam pewność.

Naprawdę nie wiem co wtedy poszło nie tak i jest to nurtujące. Jedyną "odchyłką" od normy jak mogłem popełnić w całym procesie, to użycie innego programu do flashowania. NodeMCU-PyFlasher-4.0-x86.exe coś mi nie chciał działać z początku, więc użyłem ESP8266Flasher.exe. Ale używałem go zarówno wtedy, jak i teraz (jak również do wgrania ESP Easy).

Jestem początkujący w te klocki (esp, flashowanie, kabelki) i chodzę trochę po omacku,
wybaczcie jeśli robię jakieś głupie rzeczy.

tnowak, 2020-04-09:
A tak przy okazji, to zgłaszam jeszcze "czeski błąd" z tymi placeholderami:

[image]

Zauważyłem dopiero jak zobaczyłem:

[image]

:)

bfaliszek, 2020-04-09:
Faktycznie odwrotnie wpisałem w domyślnych ustawieniach. W każdym razie to można samemu edytować, więc nie jet to nic poważnego.

Cieszę się, że problem z BME280 został rozwiązany :)

bfaliszek, 2020-04-20:
@tnowak dodałem opcję wyłączenia tego '/' na początku topicu oraz poprawiłem ten błąd z ciśnieniem i wilgotnością. Na razie w wersji prerelease dla ESP8266 z miernikiem PMS
https://github.com/hackerspace-silesia/Smogomierz/tree/dev/firmware/ESP8266

tnowak, 2020-04-21:
Chciałem wypróbować, ale wyskakuje mi błąd:
Update error: ERROR[10]: Magic byte is wrong, not 0xE9

bfaliszek, 2020-04-21:
Masz ESP32 czy ESP8266? Wersja dla ESP32 jest tutaj - https://github.com/hackerspace-silesia/Smogomierz/tree/dev/firmware/ESP32

tnowak, 2020-04-21:
Mam ESP8266, próbowałem .bin dla tego właśnie

bfaliszek, 2020-04-21:
Spróbuj pobrać jeszcze raz .bin i wgrać. Jeśli problem będzie się powtarzał, to sprawdź czy zmiana zasilacza coś da. Znalazłem info, że ten błąd może być spowodowany za słabym zasilaczem.

tnowak, 2020-04-22:
Zasilacz mam klasyczny beczkowy DC 5V 2A, więc raczej prądu pod dostatkiem.
Ale pobrałem jeszcze raz ten .bin i rzeczywiście - tym razem weszło!
Dziwności... ale najważniejsze że działa 💪

bfaliszek, 2020-04-22:
Czyli jak rozumie, chodziło o takie coś, żeby nie było '/' na początku i teraz jest już ok?

tnowak, 2020-04-22:
Tak, tzn. to załatwia większą część sprawy. Ale rzecze mówiąc byłoby już tak zupełnie idealnie, gdybyś dodał jeszcze jedną, analogiczną opcję odnośnie slasha na końcu, tj:

Add '/' at the beginning of the topic: [No]
**Add '/' at the end of the topic: [No]**    <-------- o to mi chodzi

Bo przeznaczeniem slasha w tematach MQTT jest separacja członów tematu, więc zarówno rozpoczynanie tematu od slasha jak i kończenie slashem wydaje się bezzasadne (nie ma od czego separować). Na około 90 pomiarów jakie zbieram przez MQTT żadnej temat nie kończy się slashem, tu przykładowo parę:

```
conbee/Aqara_Temp_Sensor_3H/humidity=40.37
conbee/Aqara_Temp_Sensor_3T/temperature=25.85
iobroker/mihome-humidifier/0/info/humidity=49
mqttmiot/airpurifier/humidity/state=43
mqttmiot/airpurifier/pm25/state=17
espeasy/MH-Z19B/PPM=982
espeasy/MH-Z19B/Temperature=27.0

```

W smogly jest obecnie tak (sam pewnie przyznasz, że trochę dziwnie):
```
smogly/sensor/temperature/=15.05
smogly/sensor/humidity/=37.55
smogly/sensor/pressure/=1028.61
smogly/sensor/PM10/=24
smogly/sensor/PM2.5/=23
itd.

```
A w mojej ocenie domyślnie mogłoby to wyglądać tak:
```
smogly/sensor/temperature=15.05
smogly/sensor/humidity=37.55
smogly/sensor/pressure=1028.61
smogly/sensor/PM10=24
smogly/sensor/PM2.5=23
itd.
```

Chyba lepiej pasowałoby to do praktyk stosowanych w branży i przez inne urządzenia.
I do tego wystarczy jedynie dodanie tej drugiej opcji (która jednocześnie pozwoli zachować kompatybilność wsteczną)

bfaliszek, 2020-04-22:
Dodałem 😉
Nowe pliki .bin są tutaj - https://github.com/hackerspace-silesia/Smogomierz/tree/dev/firmware

tnowak, 2020-04-22:
Działa jak złoto! Dzięki!!! 🙌 Zamykam issue bo wszystko z niej już rozwiązane!

closed 2020-04-22 by tnowak