hackerspace-silesia/Smogomierz #92: Połączenie z InfluxDB po HTTPS

ravicious, 2020-04-14:
Postawiliśmy sobie ostatnio czujnik w mieszkaniu, podpiąłem go pod InfluxDB i super jest mieć wgląd w trendy temperatury i wilgotności!

Nie udało mi się zrobić jednej rzeczy: wysyłać danych do Influxa po HTTPS. Mamy wersję 2.7.5, próbowałem dodać `https://` przed "InfluxDB database address", zmienić port na 443, ale wygląda na to, że czujnik i tak zawsze próbuje wysyłać requesty po HTTP, przynajmniej tyle wnioskuję z logów nginxa.

Nie mam pojęcia, ile wysiłku wymaga dodanie obsługi HTTPS na takim hardwarze, ale byłoby super, gdyby dało się bezpiecznie wysyłać dane do Influxa stojącego poza siecią domową.

bfaliszek, 2020-04-14:
Robiłem kilka podejść do HTTPS na ESP8266 i zawsze kończyło się to brakiem wystarczającej pamięci. Pewnie sprawdzę to znowu za jakiś czas, bo czasami robią jakieś aktualizacje w esp8266 arduino core w tym kierunku. Na chwilę obecną stabilnie działa tylko http. Tak samo jak w przypadku automatycznych aktualizacji oprogramowania. HTTPS bez problemu działa na ESP32.

galuszkak, 2020-07-14:
@bfaliszek tu jest opis z działającym HTTPS dla InfluxDB:

https://github.com/tobiasschuerg/InfluxDB-Client-for-Arduino#write-options

Generalnie certyfikat musi być dodany do kodu. W opisie twierdzą, że jest wsparcie dla ESP8266.

Jakby coś mogę pomóc w implementacji ;)

bfaliszek, 2020-07-14:
Możliwe, że dodali już wsparcie, bo wszystkie osatatnie aktualizacje dla esp8266 były związane z obsługą https. 
Ostatnio u mnie krucho z czasem, dlatego tego jeszcze nie sprawdzałem

closed 2021-11-16 by bfaliszek