hackerspace-silesia/Smogomierz #93: no matching function for call to 'esp8266webserver - problem z kompilacją kodu Smogomierz

Ralphie71, 2020-04-16:
Witam,

Chciałbym zmienić tło w stronie WWW wyświetlającej wyniki pomiarów z białego na czarne i litery białe dla kontrastu  ( wyświetlam wyniki na urządzeniu mobilnym w sposób ciągły i ekran szybko pochłania energię i rozładowuje baterie ).
Zacząłem od instalacji najnowszego 1.8.12 IDE Arduino i bibliotek dla płytek aby skompilować kod źródłowy Smogomierz i niestety jeszcze bez wprowadzania zmian w kodzie obecna wersja Smogomierza 2.7.5  wyrzuca błąd podczas kompilacji w sekcji dot. AsyncWebServer:

exit status 1
no matching function for call to 'esp8266webserver::ESP8266WebServerTemplate<WiFiServer>::on(const char [2], WebRequestMethod, void (&)())'

Czytałem o tym w Internecie i chodzi chyba o błędne użycie funkcji wymagającej parametru bez możliwości jej przekazania dalej  -> niestety nie jest to  na moim poziomie   :-(   i  nie jestem w stanie tego poprawić w kodzie ESPWebServera.

Proszę o poprawienie problemu z kodem tak aby kod źródłowy mógł się kompilować w IDE.
Z góry dziękuję.

W załączeniu kopia opisu błędu.

[blad_webserver.txt](https://github.com/hackerspace-silesia/Smogomierz/files/4487296/blad_webserver.txt)

Pozdrawiam, dziękuję  i gratuluję takiego przydatnego urządzenia jak Smogomierz.

Rafał

bfaliszek, 2020-04-16:
Pobierz jeszcze raz kod z githuba i zainstaluj wszystkie wymagane biblioteki: ArduinoJson, Adafruit BMP280 Library, DHT sensor library by Adafruit, ThingSpeak, Adafruit Unified Sensor oraz PubSubClient. Na koniec ręcznie doinstaluj biblioteki z plików zip:  ESPAsyncWebServer oraz ESPAsyncTCP. W przypadku ESP32 musimy doinstalować jeszcze AsyncTCP. W przypadku ESP32 musimy również doinstalować Pythona3 oraz pyserial. Więcej info na https://github.com/hackerspace-silesia/Smogomierz/blob/master/instrukcje/software.md#arduino---konfiguracja

Ralphie71, 2020-04-16:
Zrobiłem jak pisałeś. Niestety nie pomogło - w starym pliku 2.7.5  który opisałem powyżej pozostaje ten sam błąd a w nowym który ściągnąłem z githuba jak radziłeś jest:

Błąd kompilacji dla płytki NodeMCU 1.0 (ESP-12E Module)

zmieniałem typy płytek i dla każdej był ten sam błąd kompilacji danej płytki.

Nie wiem - pewnie coś z moim środowiskiem IDE jest nie tak bo skoro innym się kompiluje a u mnie nie to cóż.....
Prosiłbym może abyś dodał  "czarny Theme" strony www Smogomierza jako opcję dla oszczędzających energię baterii gdy będziesz przygotowywał nową paczkę aktualizacyjną ?  
Byłbym wdzięczny.

Rafał
[blad_details.txt](https://github.com/hackerspace-silesia/Smogomierz/files/4488875/blad_details.txt)

bfaliszek, 2020-04-17:
Będę pamiętał przy kolejnej aktualizacji. 
Jeśli chodzi o ustawienia ArduinoIDE:
[image]
Możesz sprawdzić jeszcze kod z brancha dev. Tam dodaję na bierząco poprawki i jak się nazbiera więcej to puszczam jako finalna wersja - https://github.com/hackerspace-silesia/Smogomierz/tree/dev

majstercar, 2020-04-17:
Fajnie - ale mam problem z esp32 pomimo zmiany nazwy sprawdzone w de-bug moja sieć widzi dalej espresif - co robię źle  wszystko działa?

bfaliszek, 2020-04-18:
Wszędzie wcześniej pisałeś, że błąd to "Błąd kompilacji dla płytki NodeMCU 1.0 (ESP-12E Module)". Jeśli masz ustawione NodeMCU, to masz ESP8266, a nie ESP32. Dla ESP32 musisz mieć konfigurację dla ESP32. 
[image]

Ralphie71, 2020-04-19:
Cześć,  
wpadłem na jeszcze jeden pomysł ->  żeby kompilować zmieniając za każdym niepowodzeniem wersję biblioteki płytek  ESP8266 Community gdyż przy najnowszej która zainstalowała się razem z Arduino był błąd kompilacji dla płytki NodeMCU 1.0 (ESP-12E Module)  jak pisałem wyżej.

Zacząłem od wersji 2.3.0  i szedłem w górę i przy wersji 2.5.0  udało mi się skompilować kod Smogomierza !!!.

Musisz chyba podawać przy plikach źródłowych Smogomierza przy której wersji biblioteki płytek tworzysz kod bo najwyraźniej wersja nowsza biblioteki nie zawsze jest zgodna wstecznie..... i stąd może być błąd kompilacji NodeMCU 1.0 (ESP-12E Module). 

Mi udało się przy 2.5.0  i już nie próbowałem kolejnej wersji żeby nie zepsuć   :-)

bfaliszek, 2020-04-20:
Zawsze używam najnowszej dostępnej, obecnie 2.6.3. W branży dev, wszystko działa i się kompiluje.

closed 2020-04-26 by bfaliszek