hackerspace-silesia/Smogomierz #96: Konfiguracja Smogly

grucha532, 2020-04-28:
Witam,

Mam problem z konfiguracją bądź odczytami z czujników. Kombinuje już drugi dzień i niestety bez powodzenia.

Otóż w konfiguratorze IP_Smogly odczyty pogody oraz jakości powietrza są następujące:
### Weather:
Temperature: 2147483647 °C
Humidity: 2147483647 %
Dewpoint: 2147483647 °C
### Air pollution:
PM1: -1 µg/m³
PM2.5:  -1 µg/m³
PM10:  -1 µg/m³

Sprawdziłem 3x i wszystko podpięte jest poprawnie. Użyłem czujników PMS7003 oraz DHT22 (podpięte do ESP8266). W ustawieniach urządzeń Smogly ten pierwszy ustawiony jako PIN TX: D1 oraz PIN RX: D2, natomiast ten drugi DHT22 PIN SDA: D7 oraz PIN SCL: D7. W rzeczywistości jednak czujnik ten ma tylko 3 wyjscia, dlatego nie wiedziałem co wybrać jako SCL. Probówałem tutaj podpinać czujnik DHT22 zarówno z rezystorem podciągającym (10k) jak i bez niego. W obu przypadkach te same wyniki.
Podpiąłem również Homebridge lecz tam czujniki te chociaż się wyświetlają to nie odpowiadają.

Czy miał ktoś podobny problem i/lub wie jak to rozwiązać?

Z góry dziękuję za rady/wskazówki.

bfaliszek, 2020-04-28:
Sprawdź jak będzie działać bez DHT22

grucha532, 2020-04-30:
Nie wiem co ostatecznie miało wpływ, lecz czujnik pyłów zaczął prawie działać tj. czasem pokazuje wartości PM a czasem 0µg/m³ i dopiero po kilku minutach się aktualizuje (co i tak jest postępem). Obecnie działa z podpiętym DHT22, którego wyniki to nadal:
Temperature: 2147483647 °C
Humidity: 2147483647 %
Dewpoint: 2147483647 °C

Mam dwa czujniki DHT22 i testowałem oba i oba działają - sprawdzone na RPi), tak jak pisałem zarówno z rezystorem jak i bez niego. Czy ustawienia Smogly mam poprawne? Jeśli chodzi o SCL to próbowałem wszystkich możliwości.

Nie wiem czy to efekt czy może wskazówka szukając przyczyny, lecz komunikacja z Homebridge (tj. wyników w aplikacji Home) jest słaba tj. aby na smartphonie sprawdzić jakość powietrza muszę kliknąć kafelek "Air Quality" i wtedy albo się zaktualizuje i pokaże ocenę np. "good", po czym wychodzę z tego jednego czujnika na stronę pokoju w aplikacji gdzie czujników jest wiele, w tym momencie wszystkie czujniki się aktualizują a następnie kafelek "Air Quality" zmienia status na brak odpowiedzi. --> jeśli to jest jedynie negatywny efekt a nie wskazanie, że zawodzi tutaj gdzieś komunikacja to wrócę do niego później - po rozwiązaniu problemu z czujnikiem.

bfaliszek, 2020-04-30:
Naprawiłem obsługę DHT22. Wrzuć sobie .bin prerelease z https://github.com/hackerspace-silesia/Smogomierz/tree/dev/firmware/ESP8266 Wersja dla ESP8266 i miernika PMSx003.

grucha532, 2020-04-30:
Działa! Super, dziękuję :)

bfaliszek, 2020-05-03:
https://github.com/hackerspace-silesia/Smogomierz/releases/tag/v2.7.9

closed 2020-05-03 by bfaliszek