hackerspace-silesia/Smogomierz #99: Zewnętrzna antena Wifi, jakieś pomysły?

tnowak, 2020-05-31:
Hej,

Okazuje się, że mam na balkonie dość słaby zasięg Wifi. Tzn. sięga, ale w moim bloku jest b. dużo AP na 2,4GHz i jest śmietnik w eterze w tej częstotliwości. Tak czy siak RSSI oscyluję mi w okolicach -70 dBm i zdarzają się nawet sytuacje, w których urządzenie jest niedostępne.

Wstępnie rozważam dwie opcje: dołożyć Unifi mesha na balkonie, ale to trochę overkill, albo... no właśnie i tu pytanie. Czy ktoś dokładał zewnętrzną antenę do ESP na NodeMCU v3? Da się? Duża rzeźba?

bfaliszek, 2020-05-31:
ESP8266 ESP-07 ma złącze antenowe U.FL.

tnowak, 2020-05-31:
Uhm. Czyli co, musiałbym wylutować z NodeMCU v3 to co tam jest i wlutować taki ESP-07 tak?

tnowak, 2020-05-31:
To jeszcze jedno pytanie. Jakbym kupił Wemos D1 mino Pro taki jak tutaj:
https://allegro.pl/oferta/wemos-d1-mini-pro-16m-v1-0-8309943224
to smogly na nim bez problemu ruszy?

bfaliszek, 2020-05-31:
Powinien działać bez problemu, tylko pewnie będziesz musiał samemu skompilować kod, bo ten który jest dla esp8266 jest dla wersji 4mb

bfaliszek, 2020-05-31:
> Uhm. Czyli co, musiałbym wylutować z NodeMCU v3 to co tam jest i wlutować taki ESP-07 tak?

Nie, takie coś nie zadziała. ESP8266 mają różne wersja. NodeMCU używa ESP-12. Jest też np. ESP-01, które ma tylko 2 piny gpio. ESP-07 ma za to zewnętrzną antenę. Każde z nich to esp8266 i będzie na nich działał kod smogomierza.

tnowak, 2020-05-31:
Ok, czyli realnie zostaje mi tylko opcja tego D1 Mini Pro 16m v1.0 i samodzielnie skomplikowanie kodu. Jakieś wskazówki dla osoby, która nigdy tego nie robiła? Coś w stylu biorę Arduino IDE, wczytuje source i klikam "kompiluj"? :)

closed 2021-11-16 by bfaliszek