hackerspace-silesia/Smogomierz #101: Automatyczna rejestracja stacji w API AirMonitor

airmonitor, 2020-06-15:
Cześć,

Jak widzisz aby dodać automatyczną rejestracje stacji Smogomierza w API AirMonitora gdy zostanie zaznaczona opcja do wysyłki danych do AirMonitora?

Teraz jest formularz, natomiast chyba czas zmienić podejście na bardziej automagiczne?

POST z treścią:
```json
{"particle_sensor_model": "SDS021", "latitude": "55.0", "longitude": "16.0", "email": "name.surname@gmail.com", "temperature_sensor_model": "BME280", "humidity_sensor_model": "BME280", "pressure_sensor_model": "BME280", "carbon_dioxide_sensor_model": "MH-Z19", "tvoc_sensor_model": "CCS811"}

```

bfaliszek, 2020-06-16:
Teraz JSON wygląda na przykład tak:
```json
{
"lat" = "55.0",
"long" = "16.0", 
"pm1" = "10.0", 
"pm25" = "15.0", 
"pm10" = "20.0", 
"sensor" = "PMS7003"
}
```
lub w przypadku pomiarów temperatury/wilgotności/ciśneinia:
```json
{
"lat" = "55.0",
"long" = "16.0", 
"pressure" = "1000",
"temperature" = "23",
"humidity" = "66",
"sensor" = "BME280"
}
```
To pozostaje bez zmian i tylko za pierwszym razem wysłać raz POST w takiej formie jak podałeś?

airmonitor, 2020-06-17:
powyższy JSON z pomiarami możesz też lekko zmienić aby wartości metryk przesyłanych przez sensory były wartością float - zmienno przecinkową do z maksymalnie dwoma miejscami po przecinku.

Teraz jest string - "1000" po zmianach 1000.00

Natomiast zapomniałem dodać, że url do rejestracji stacji to

https://apigw.airmonitor.pl/station - metoda http POST

airmonitor, 2020-06-29:
Kontynuując temat:

url do rejestracji - https://airmonitor.pl/prod/station [POST]
headers= {"X-Api-Key: "..."} - podeślę prywatnie :)

w odpowiedzi przychodzi json:

{
    "status": "Station registered",
    "api_key": "fethderq3t4whtedr452asDAHRTASDFDGG"
}

api_key ma długość 42 znaków

api_key należy następnie zapisać na stacji i wykorzystać za każdym razem gdy będzie wysyłany pomiar do api.
Wysyłany pomiar musi mieć header o treści:

{"X-Api-Key: "fethderq3t4whtedr452asDAHRTASDFDGG"}} - jest to wartość pobrana przy rejestracji stacji.

URL do wysyłania pomiarów:
https://airmonitor.pl/prod/measurements [POST]

Poniżej przykład jsona ale to już znasz :)
{"lat": "52.162300", "long": "21.228300", "pm1": 5, "pm25": 7, "pm10": 7, "sensor": "PMS7003", "time": "2020-06-14T15:34:19Z"}

aerorafik, 2020-08-17:
Heja,
korzystając z wątku airmonitor , gdzie należy wpisać API key aby dane były przyjęte przez serwis ?
Rejestracja manualna , API key pozyskane. Teraz kwestia konfiguracji smogomierza.

Dziękuję z góry

airmonitor, 2020-08-18:
Cześć.

Tak z grubsza to dla kilku języków masz listingi kodu tutaj - https://airmonitor.pl/prod/api

Natomiast szybka odpowiedź to nagłowek http.

Jako, że jestem pythonowy to poniżej przykład kodu

```
import json  
import requests  
data = {  
    "lat": str(55.000),  
    "long": str(15.000),  
    "pm25": 25,  
    "pm10": 10,  
    "sensor": "SDS021"  
    }  
url = 'https://airmonitor.pl/prod/measurements'  
resp = requests.post(url, timeout=10, data=json.dumps(data), headers={"X-Api-Key": uzyskany_klucz_api, "Content-Type": "application/json"})  
resp.status_code

```

zwróć uwagę na X-Api-Key

airmonitor, 2020-11-29:
Cześć.

Fajnie, że to już zaimplementowane. Planuję na koniec roku mały przestój legacy api stąd też Ci, którzy zaczęli korzystać z https://airmonitor.pl/prod/measurements nie zauważą brakujących pomiarów w czasie planowanego maitenance :)

hanskiepl, 2021-01-08:
Cześć, nie do końca rozumiem te wszystkie pythony i inne :-) ale jak mam smogomierz, zarejestrowałm się przez formularz współrzędnymi takimi samymi jak wpisałem w smogomierzu (z tą różnica że w jednym przypadku podełm kropki a w drugim przecinek w tych współrzędnych zgodnie z instrukcją) to ten klucz API który dostałem zwrotnie w zasadzie ni ejest mi do niczego potrzebny ? airmonitor będzie mnie automatycznie rozpoznawał po współrzędnych ?
Pytam, gdyż nie ma nigdzie możliwości wprowadzenia tego klucza w smogomierzu - albo przynajmniej ja tego nie odnajduję mimo skrupulatnej weryfikacji.

airmonitor, 2021-01-08:
> Cześć, nie do końca rozumiem te wszystkie pythony i inne :-) ale jak mam smogomierz, zarejestrowałm się przez formularz współrzędnymi takimi samymi jak wpisałem w smogomierzu (z tą różnica że w [...]

Cześć.

Smogomierz korzysta ze starego API, które może za jakiś czas przestać działać gdyż nie jest to rozwiązanie które elastycznie skaluje się względem ilości przesyłanych danych to raz, nie jest fault tolerant i nie jest to HA.

Tak więc jeśli kiedyś przestanie działać a z pewnością przestanie skoro smogomierz nie jest aktualizowany i to API zostanie przeciżone.. :/

Już dawno temu prosiłem aby przepiąć na nowe API do którego jest właśnie potrzebny klucz, a to nowe API ma:
- HA
- fault tolerant
- przyjmuje do 10 000 RPS (request per second)
Więc mówimy tutaj o zupełnie innej klasie API w porównaniu do starej wersji...

airmonitor, 2021-01-09:
@hanskiepl coś mi się wydaje, że smogomierz doszedł do etapu gdy jego rozwój musi być poparty finansowaniem

https://hs-silesia.pl/smogomierz/

hanskiepl, 2021-01-09:
Może to nie jest zły pomysł. Szkoda tylko, że nie jest zbyt widoczna ta akcja wsparcia ich działaności.
Wierze jednak, że ktoś to będzei ciągnął nadal. Chyba, że jest jaka inna alternatywa ?

bfaliszek, 2021-01-10:
W nowej wersji zaktualizowałem tylko używane biblioteki. Największym problemem jest u mnie brak czasu. Musiałbym mieć przynajmniej kilka a najlepiej kilkanaście dni żeby ogarnąć wszystkie zaległe rzeczy w projekcie, a fizycznie nie mam na to teraz czasu i możliwość. 
Dlatego zachęcam do robienia PRów, to je dodam do kodu Smogomierza.

airmonitor, 2021-01-10:
@hanskiepl 

Uważam, że brak czasu Błażeja ma to samo źródło jak wiele innych polskich projektów - brak angażu społeczności dookoła projektu. Samo wrzucanie PR'ek, że fajnie byłoby mieć to i tamto nie jest stricte wsparciem. Potrzeba programistów, testerów, którzy chcą kodzić i testować. Tego nie ma, a oczekiwania każdy ma takie, że chce za darmo i musi działać z jego sensorami :)

Anyway, aby nie kontynuować zbytnio wywodów filozoficznych, alternatywa jest ale zdecydowanie bez takiego ładnego gui jakie Błażej stworzył.

https://airmonitor.pl/prod/micropython

Wyszedłem z założenia, że stacja ma wyłącznie monitorować parametry środowiska i wysyłać dane - nic mniej, nic więcej.

Oddzielny front-end ma prezentować dane, a osobny backend ma odpowiadać za wszystko inne.

bfaliszek, 2021-04-08:
Dodałem obsługę kluczy API przy wysyłce danych do AirMonitora - https://github.com/hackerspace-silesia/Smogomierz/commit/0afc98bac432b8783c8884e2efadc8af4fc8b5bd#diff-cd91ff19b52817b69a30311a4381bfe4c3b72cf7fbcd1ff21224cc523a48ab34

closed 2021-11-16 by bfaliszek