hackerspace-silesia/Smogomierz #102: Padł PMS?

tnowak, 2020-06-17:
Zanim napiszę do Botlandu w sprawie reklamacji, chciałem się tu upewnić czy dobrze wszystko widzę.
Po ok 2.5 miesiącach dobrej pracy wygląda na to ze padł mi czujnik PMS.
Cały czas pokazuje pomiary 0. Tak to wygląda w smogly:

[image]

Co dokładnie znaczy 0 w kolorze cyan?
Czy te wykresy na dole oznaczają, że on coś tam jednak mierzy? (niby są wartości pomiędzy 0 i 1.

Choć powietrze jest ostanio dobrej jakości, to jednak nie dałem wiary że non stop doskonałej. Wziąłem czujnik do domu i rozpyliłem jakiś aerozol w pobliżu. Czujnik PM w oczyszczaczu powietrza poleciał w kosmos, a PMS - nic.

Popsuł się, reklamować?

bfaliszek, 2020-06-17:
Jeśli nawet w domu przy aerozolu lub dymie ze świeczki/zapałki wciąż pokazuje 0, to chyba nie działa.
Kolory są uzależnione od poziomu PM, jeśli wartość wynosi 0, to oznacza, że jest wyświetlana w kolorze idealnie czystego powietrza ;)

Przed wysłaniem możesz jeszcze sprawdzić czy wszystko ok z kablami z PMS. Czasami z nimi jest problem.

tnowak, 2020-06-17:
Błażej, dziękuję Ci za jak zwykle celne rady. Miałeś rację - najwyraźniej był jakiś problemik z kabelkami. Otworzyłem obudowę, wpiąłem wszystkie jeszcze raz i:

[image]

:)

closed 2020-06-17 by bfaliszek