hackerspace-silesia/Smogomierz #107: metrics.airmonitor.pl - Witryna nie istnieje

jarekera, 2020-11-20:
Witam, od dłuższego czasu na stronie smogomierza do serwisów Airminitor jest komunikat:
"Nie udało się nawiązać połączenia
Firefox nie może nawiązać połączenia z serwerem metrics.airmonitor.pl:3000. 
"
były już aktualizacje od czasu wystąpienia, ale problem jest w dalszym ciągu.
Na stronie Airmonitor dane pokazują się poprawnie, ale widzę że chyba zmienił się adres strony prezentujący te mierniki brak numeru portu, czy możecie to poprawić ?

bfaliszek, 2020-11-20:
Gdy tylko znajdę chwilę czasu. Ewentualnie możesz zrobić PR z tą zmianą.

jarekera, 2020-11-23:
Aż tak oblatany w tym nie jestem. Ok poczekam jak będzie poprawka jakaś.

galuszkak, 2020-11-27:
@bfaliszek @jarekera wrzuciłem PRa z poprawką w #108

jankes46, 2020-12-02:
a jak te poprawki wgrać ?
muszę podłączyć arduino po kablu czy da się po wifi ?
przepraszam za takie pytania ale dopiero zaczynam zabawę z Arduino

bfaliszek, 2020-12-02:
@jankes46 jeśli masz urządzenie oparte o ESP8266 i miernik PMS7003 lub PMS5003 to możesz wgrać przez przeglądarkę plik .bin. Właśnie wrzuciłem skompilowaną wersję z poprawką dodaną przez @galuszkak - https://github.com/hackerspace-silesia/Smogomierz/blob/dev/firmware/ESP8266/prerelease_ESP8266_Smogomierz_2.8.0_PMS_build_2.12.2020.bin

jarekera, 2020-12-02:
Dzięki działa

jankes46, 2020-12-02:
A jeszcze jedno czy na stronie powinienem widzieć wykres czy to będzie za jakiś czas od uruchomienia ?
[image]

jarekera, 2020-12-02:
On uśrednia wyniki i pojawia się z opóźnieniem, ale jak poprawnie skonfigurowałeś cos takiego zobaczyć na stronie airmonitor - [https://metrics.airmonitor.pl/d/000000002/airmonitor?orgId=1&var-latitude=52.315742&var-longitude=21.138281&from=now-24h&to=now&refresh=10s](url)

jankes46, 2020-12-02:
jest tak:
https://metrics.airmonitor.pl/d/000000002/airmonitor?orgId=1&from=now-24h&to=now&refresh=10s&var-latitude=52.315742&var-longitude=19.433179

jarekera, 2020-12-03:
Wyglada ze nie wysyłasz danych do Airmonitor przejrzyj jeszcze ustawienia i
sprawdź czy się zarejestrowałeś w Airmonitor.

<goog_1659957815>
Instrukcja -
https://github.com/hackerspace-silesia/Smogomierz/blob/master/instrukcje/software-additionals.md

[image: obraz.png]

Pozdrawiam JZ

artur5236, 2021-01-19:
> @jankes46 jeśli masz urządzenie oparte o ESP8266 i miernik PMS7003 lub PMS5003 to możesz wgrać przez przeglądarkę plik .bin. Właśnie wrzuciłem skompilowaną wersję z poprawką dodaną przez @galuszkak [...]

Link nie działa. 
Ja po uruchomieniu Smogly ma coś takiego:
[image]

bfaliszek, 2021-01-19:
Link nie działa, to jest nowszy build - https://github.com/hackerspace-silesia/Smogomierz/tree/dev/firmware/ESP8266

closed 2021-04-08 by bfaliszek