hackerspace-silesia/Smogomierz #109: Problem z kompilacją na ESP32

ebuke, 2020-12-16:
cześć,
Mam problem z kompilacją na płytce ESP32. Ktoś ma pomysł o co chodzi? 

wysypuje się na linii 875
#elif defined ARDUINO_ARCH_ESP32
    sprintf(device_name, "Smogly-%06X", ESP.getEfuseMac());

exit status 1
comparison with string literal results in unspecified behaviour [-Werror=address]

bfaliszek, 2020-12-17:
Hej! 
Użyłeś kodu z dev? https://github.com/hackerspace-silesia/Smogomierz/tree/dev
Zmieniałeś coś w nim? Jaką masz ustawioną płytkę ESP32 w ArduinoIDE? Ja kompiluję zawsze na ESP32 Dev Module

ebuke, 2020-12-17:
Hej. Kod mam z mastera. Nic nie zmieniałem. Kompiluję też na ESP32 DEV Module. Może brakuje mi którejś biblioteki poza tymi które są w instrukcji?

ebuke, 2020-12-19:
Winne ustawienia ostrzeżeń kompilatora. Były za wysokie.

closed 2020-12-19 by ebuke