hackerspace-silesia/Smogomierz #113: Guru Meditation Error: Core  1 panic'ed (LoadProhibited) podczas sprawdzania update

mjarosie, 2021-03-01:
Hej, dzięki za super robotę, instrukcje Smogomierza są bardzo pomocne! : )

Podczas włączania urządzenia, z jakiegoś powodu zapytanie do https://smogomierz.hs-silesia.pl/firmware/latest_esp32.json zwraca kod `-1` ([tutaj](https://github.com/hackerspace-silesia/Smogomierz/blob/375a111aa2468572a48c8fa69fa649c81c447790/src/autoupdate.cpp#L117)). Kod dochodząc do [tej linii](https://github.com/hackerspace-silesia/Smogomierz/blob/375a111aa2468572a48c8fa69fa649c81c447790/src/autoupdate.cpp#L164), rzuca błąd:

```
connected...yeey :)
Waiting for NTP time sync: 
Current time: Mon Mar  1 18:53:26 2021
GET... failed, error: connection refused
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x4000159c  PS      : 0x00060c30  A0      : 0x800f224a  A1      : 0x3ffb1b30  
A2      : 0x3ffc5820  A3      : 0x00000001  A4      : 0x00000020  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x80102029  A9      : 0x3ffb1b00  
A10     : 0x3ffc5820  A11     : 0x000000ff  A12     : 0x00000029  A13     : 0x3ffb1b20  
A14     : 0x3ffb1aa0  A15     : 0x00000008  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000001  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffb  

ELF file SHA256: 0000000000000000

Backtrace: 0x4000159c:0x3ffb1b30 0x400f2247:0x3ffb1b40 0x400e4a9f:0x3ffb1f70 0x4010ad52:0x3ffb1fb0 0x4008a6f2:0x3ffb1fd0

Rebooting...
```

`strncpy` jest wywolane na niezainicjalizowanej pamięci zadeklarowanej [tutaj](https://github.com/hackerspace-silesia/Smogomierz/blob/375a111aa2468572a48c8fa69fa649c81c447790/src/autoupdate.cpp#L85).

Po użyciu [exception stack trace decoder](https://github.com/me-no-dev/EspExceptionDecoder) register dump wygląda tak:

```
PC: 0x4000159c
EXCVADDR: 0x00000001

Decoding stack results
0x400f2247: checkUpdate(int) at /var/folders/7d/0fcl4nds6xd1_sjw8c0ndsn80000gp/T/arduino_build_272800/sketch/src/autoupdate.cpp line 164
0x400e4a9f: setup() at /Users/mjarosie/source/mjarosie/Smogomierz/Smogomierz.ino line 660
0x4010ad52: cbuf::write(char const*, unsigned int) at /Users/mjarosie/Library/Arduino15/packages/esp32/hardware/esp32/1.0.5/cores/esp32/cbuf.cpp line 171
0x4008a6f2: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143
```

mjarosie, 2021-03-01:
Quickfix: zainicjalizowanie zmiennej, np:

```
ServerSW = "2.7.9";
```

bfaliszek, 2021-03-02:
https://github.com/hackerspace-silesia/Smogomierz/commit/ed85019f8a8f7bc6c4407df713f4310df3992f6a

closed 2021-04-08 by bfaliszek