hackerspace-silesia/Smogomierz #116: Podanie o dodanie SHT-30

tnowak, 2021-06-02:
Cześć!

Fajnie że Sensirion SPS-30 został dodany.
Czy jest szansa na dodanie też czujnika temperatury i wilgotności Sensiriona SHT-30?
Ostatnio natknąłem się na taki fajny w obudowie pyłoszczelnej:
https://botland.com.pl/czujniki-pogodowe/14969-sht-30-czujnik-temperatury-i-wilgotnosci-odporny-na-warunki-atmosferyczne-i2c-1m-adafruit-4099.html
i myślę że fajnie, gdyby tez był obsługiwany

Pozdro!

bfaliszek, 2021-06-03:
Trochę drogie rozwiązanie jak na SHT-30, zwłaszcza, że widziałem je za ~40zł - https://allegro.pl/oferta/czujnik-temperatury-i-wilgotnosci-gleby-sht30-10092884720 

Postaram się dodać obsługę SHT-30 przy kolejnej aktualizacji

tnowak, 2021-08-27:
Hej i jak tam Twoje możliwości czasowe?
Bo sprzęt leży już na półce i czeka na testy

closed 2021-11-16 by bfaliszek