hackerspace-silesia/Smogomierz #117: SPS30 Sensirion nie wyłącza się wiatrak w stanie uśpienia.

austinroolon, 2021-08-06:
Podczas stanu uśpienia nie wyłącza się wiatrak SPS30 Sensirion. Podobnie jest między pomiarami. Niepotrzebnie się kręci, wzrasta zużycie energii. Bardzo dokładnie wyjaśnia obsługę LOW POWER nota producenta. Szkoda ,ze nie zaimplementowano jej.
https://www.sensirion.com/fileadmin/user_upload/customers/sensirion/Dokumente/9.6_Particulate_Matter/Application_Notes/Sensirion_Particulate_Matter_AppNotes_SPS30_Low_Power_Operation_D1.pdf

BTW. nie wiem gdzie zadać pytanie o kompilację pod ESP32 wersji z obsługą SPS30, BMP280 i HTU21d. Który kod .ino pobrać? Może przygotowałbym obsługę LPO dla SPS30

bfaliszek, 2021-08-07:
Z SPS30 mam taki problem, że mam mega starą wersję tego miernika i nie działa w nim sporo funkcji. W tym właśnie usypianie, dlatego ciężko mi to przetestować. Najnowszy kod zawsze jest w branchu dei - https://github.com/hackerspace-silesia/Smogomierz/tree/dev a obsługiwany czujnik pyłu wybiera się tutaj - https://github.com/hackerspace-silesia/Smogomierz/blob/dev/Smogomierz.ino#L19-L23 Jeśli chodzi o samo ESP32, to po prostu wybierasz je w ustawieniach ArduinoIDE

austinroolon, 2021-08-09:
Jest w kodzie zaimplementowane usypianie SPS30? 
ESP8266WebServer.h  działa z ESP32? ta biblioteka jest użyta w kodzie.

Eksperymentalnie robię drugi Smogomierz z akumulatorem ładowanym z panela słonecznego. Zrobiłem na szybko z pomiarem HTU21, BMP280 (który sam z siebie padł..), pomiar natężenia światła BH1750, i analogowo z A0 mierzę poziom zapylenia analogowym Sharpem GP2Y1014AU0F + wentylator. Skaliborwałem go z SPS30 i jest nieźle. Pomiary są wiarygodne, choć bez rozróżniania wielkości cząstek.  ESP32 ma chyba stabilniejsze napięcie odniesienia do ADC? Chcę jeszcze wdrożyć pomiar napięcia akumulatora. Wszystko lecieć ma do ThinkSpeak co 15 min. Głębokie usypianie pomiędzy. Również czujników. Niedługo wrzucam czujniki do kodu Waszego smogomierza.  Potem może na Azure. Dzięki za całą deweloperkę.

bfaliszek, 2021-08-09:
Wybudzanie: https://github.com/hackerspace-silesia/Smogomierz/blob/dev/Smogomierz.ino#L2222
Usypianie: https://github.com/hackerspace-silesia/Smogomierz/blob/dev/Smogomierz.ino#L2267