hackerspace-silesia/Smogomierz #118: Problem z zasilaniem

BartekMilew, 2021-10-01:
Czy ktoś może wie w czym moze być problem? Podłączam kabel USB do ESP8266 bezpośrednio i wszystko hula. Jak podłączę przez wtyczkę micro USB jak w projekcie to nie działa. Wymieniałem już wtyczke i kabel. Nadal nic.
[image]
[image]

bfaliszek, 2021-10-01:
Sprawdź inne G na płytce ESP8266

BartekMilew, 2021-10-01:
Działa !!! Nie wiem czemu tak, ale super !

bfaliszek, 2021-10-04:
W niektórych chińskich esp8266 nie działa G przy VIN. Nie wiem z jakiego powodu, ale raz na ileś przypadków trafia mi się takie esp8266 😉

closed 2021-10-04 by bfaliszek