hackerspace-silesia/Smogomierz #150: Smogly v.2.8.1 SDS - Dziwne linie w ustawieniach urządzenia

heliolux, 2021-12-22:
W ustawieniach urządzenia zauważyłem dziwne linie zamiast dodatkowych uprawnień.

[image]

Próbowałem zrobić downgrade do wcześniejszych wersji (NodeMCU-PyFlasher z opcją 'Erease flash - Yes, wipes all data". Mimo powrotu i ustawienia opcji automatycznych aktualizacji na 'NIE' po zmianie defaultowych ustawień, oprogramowanie samo się aktualizuje. Na początku przy nowej wersji nie widać tych linii a wszystkie parametry są widoczne. Po ustawieniu parametrów i zapisie, gdy nastąpi restart widać coś jak na załączonej ilustracji.

bfaliszek, 2021-12-22:
Sprawdzę w wolnej chwili. Miałem kilka razy taki problem w wersji na PMSx003. Całość działała ok(pomiary i wysyłanie danych), tylko jedno menu się nie ładowało.

heliolux, 2021-12-22:
W 'Smogly - Ustawienia Serwisów' widzę wszystko ładnie i przycisk 'Zapisz' jest dostępny.

SajkMajk, 2022-01-11:
Też mam takie krzaki w tym miejscu. Wykorzystuję PMS7003. 
Dodatkowo nie mogę wejść w konfigurację MQTT bo też widzę krzaki albo nie widzę nic oprócz zielonego paska u góry.
[image]

Jest jakiś inny sposób, żeby skonfigurować MQTT?

eReS72, 2022-09-28:
> W ustawieniach urządzenia zauważyłem dziwne linie zamiast dodatkowych uprawnień oraz przycisku 'Zapisz'.
> 
> [image]
> 
> Próbowałem zrobić downgrade do wcześniejszych wersji (NodeMCU-PyFlasher z opcją 'Erease flash - Yes, wipes all data". Mimo powrotu i ustawienia opcji automatycznych aktualizacji na 'NIE' po zmianie defaultowych ustawień, oprogramowanie samo się aktualizuje. Na początku przy nowej wersji nie widać tych linii a wszystkie parametry są widoczne. Po ustawieniu parametrów i zapisie, gdy nastąpi restart widać coś jak na załączonej ilustracji.

Mam dokładnie tak samo z PMSx003.
ESP8266, wersja softu prerelease 2.8.2, BME280.
Wszystko wygląda dobrze do momentu zdefiniowania czujników.
Potem już jest tak jak na załączonym screenie.
Soft wgrywałem przez Arduino. Dodatkowo dochodzą problemy z zapisywaniem sieci opisane w innych wątkach.
Nie próbowałem niestety starszych wersjo softu.