hackerspace-silesia/Smogomierz #151: Bład zapisu ustawień

monteros007, 2021-12-26:
Po ostatnim update praktycznie nic nie działa prawidłowo. Prosty zapis ustawień jak zmiana wysokości, kończy się przerwaniem pracy i błędem wczytywania strony. To co idzie w URL:
http://IP/config_device_save?DEVICENAME_AUTO=yes&LANGUAGE=polish&LATITUDE=50.228&LONGITUDE=20.887&MYALTITUDE=107&THP_MODEL=BME280&DUST_MODEL=PMS7003&CONFIG_FIRST_THP_SDA=D4&CONFIG_FIRST_THP_SCL=D3&CONFIG_DUST_TX=D1&CONFIG_DUST_RX=D2&FREQUENTMEASUREMENT=no&DUST_TIME=30&NUMBEROFMEASUREMENTS=10&DEEPSLEEP_ON=no&DISPLAY_PM1=yes&CONFIG_AUTH=no&CONFIG_USERNAME=admin&CONFIG_PASSWORD=password&DEBUG=no&MODEL=white&AUTOUPDATE_ON=no&submit1=Zapisz

Kiedy już uda się coś zapisać, to nie idzie ustawić influx db, przycisk zapisz zlewa się z wpisaniem hasło do bazy.

ESP8266.bin

mrozu-poland, 2021-12-31:
Nie ma możliwości zapisu konfiguracji dla serwera MQTT, nie zapamiętuje hosta, portu, i parametrów kolejek.
Nie ma możliwości zapisu konfiguracji dla ThingSpeak-a (pola ChanellId - cały czas jest 57365, i ReadAPIKey - nie zapamiętuje). WriteApiKey zapisuje się i dane są wysyłane poprawnie do serwisu.

bryan-pl, 2022-01-01:
Dokładnie to samo mam na ESP32 - po wrzuceniu flasha przez esphome-flasher konsula zrzuca takie coś:
_[00:16:49]Mounting FS...
[00:16:49]E (65) SPIFFS: mount failed, -10025
[00:16:49]Failed to parse config file
[00:16:49]Failed to load config
[00:16:49]Saving the current config..._
Smogly działa itp, ale tak jak u Mrozu-Poland nie zapisuje ustawień MQTT i przez to MQTT nie działa.

bfaliszek, 2022-01-04:
Czyli jak rozumiem problem jest tylko ustawieniami MQTT?

monteros007, 2022-01-04:
> 

Z zapisem danych do influxDB tak samo. To w ogóle nie działa po ostatnim update.

bryan-pl, 2022-01-04:
Z dodatkowych rzeczy - niekoniecznie związanych z zapisem konfiguracji - jak zaznaczy się wyświetlanie danych z airmonitor czy community.sensor, to okienka owszem są, ale w przekazanych żądaniach HTML nie ma identyfikatora stacji / lokalizacji i wykresy są puste.

kpoow, 2022-01-04:
U mnie jest podobnie z wysypującym się urządzeniem przy zapisie ustawień (mam wgrany najnowszy soft dla ESP8266) . Dodatkowo jak próbuję połączyć płytkę z siecią Wifi to nie ładuje się UI do wyboru sieci. Da się to obejść uderzając bezpośrednio w link http://smogomierz-ip/wifisave?s=<nazwa_sieci>&p=<haslo>, wtedy poprawnie połączy się z siecią.

bfaliszek, 2022-01-04:
> Z dodatkowych rzeczy - niekoniecznie związanych z zapisem konfiguracji - jak zaznaczy się wyświetlanie danych z airmonitor czy community.sensor, to okienka owszem są, ale w przekazanych żądaniach HTML nie ma identyfikatora stacji / lokalizacji i wykresy są puste.

Nie wiem jak z airmonitor, ale wykresy z community.sensor pojawiają się po jakimś czasie. Na początku są puste. Dopiero, gdy będzie trochę danych w ich bazie, to są generowane. Nie wiem dokładnie po jakim czasie ale na pewno nie szybciej niż 30 minut. Dopiero, gdy będzie tyle danych i wykona się restart urządzenia, to będzie dostępne ID dla community.sensor

bryan-pl, 2022-01-04:
Na stronie community sensor te statystyki są od tygodnia i oba wykresy pojawiają się OK. Problem jest po stronie smogomierza, bo wyświetla puste wykresy z uwagi na to, że w linku do nich nie ma id stacji (node=0) zamiast node=id_stacji_w_community_sensor: 

> <iframe src='https://maps.sensor.community/grafana/d-solo/000000004/single-sensor-view?orgId=1&panelId=2&var-node=0' ></iframe>

Natomiast na airmonitor wykresy wyświetlają się puste błędem 404. W linku do nich są przekazane koordynaty stacji, na stronie airmonitor dane ze smogomierza są widoczne.

monteros007, 2022-01-04:
Ta wersja nie nadaje się na produkcję moim zdaniem. Jak dla mnie to w 2.7.9 poprawić CSS na poprawny URL i wydać wersję 2.7.9.1. Próbowałem sam skompilować 2.7.9 z tą poprawką ale za cienki jestem na c++ ;/

bfaliszek, 2022-01-04:
@bryan-pl 
Mówisz o wersji ESP32 czy ESP8266?
Na ESP8266 nie działa airmonitor. 

Jeśli chodzi o sensor.community, to żeby wyświetlić wykresy musisz mieć podane odpowiednie(takie same jak na ich stronie) współrzędne GPS. Urządzenie przy starcie sprawdza najbliższe stacje sensor.community dla tych współrzędnych i ściąga ich ID. To ID jest wymagane do wyświetlania wykresów. Nie wiem jak inaczej wydobyć to ID, bo jest ono inne niż to w panelu na stronie sensor.community. Ogólnie to przy wykresach sensor.community dwie rzeczy mogą pójść źle:
1) rozbierze koordynaty w ustawieniach i panelu sensor.community
2) błąd przy pobieraniu ID przy starcie Smogomierza

bfaliszek, 2022-01-04:
**MQTT**
Powrót do pełnego wsparcie MQTT w 2.8.x chcę zrobić po tym, gdy wersja dla ESP32 będzie dobrze działać z BME280/PMSx003. Właśnie skończyłem to robić i powinno być teraz ok, ale chcę to przez kilka dni potestować.
W 2.7.9 MQTT chyba działało poprawnie, ale nie korzystałem z niego, więc ciężko mi potwierdzić.

bryan-pl, 2022-01-04:
> @bryan-pl Mówisz o wersji ESP32 czy ESP8266? Na ESP8266 nie działa airmonitor.

Mówię o ESP32 gdzie nie trzyma części ustawień - szczególnie tych odnośnie przy MQTT

na ESP 8266 problemy z tą wersją są inne - każda próba zapisu ustawień powoduje reset ustawień wifi i smogomierz ustawia się na domyślnym wifi.

> Jeśli chodzi o sensor.community, to żeby wyświetlić wykresy musisz mieć podane odpowiednie(takie same jak na ich stronie) współrzędne GPS. Urządzenie przy starcie sprawdza najbliższe stacje [...]
To muszę zbadać, bo luftdaten operuje na stałoprzecinkowych liczbach przy podawaniu lokalizacji.

bfaliszek, 2022-01-04:
@bryan-pl w sumie mogę dać opcję wpisywania tego ID ręcznie. Po kliknięciu w Dane odpowiedniego urządzenia na devices.sensor.community jest Sensor ID dla miernika pyłu i on jest potrzebny do wykresów.

bryan-pl, 2022-01-04:
> @bryan-pl w sumie mogę dać opcję wpisywania tego ID ręcznie. Po kliknięciu w Dane odpowiedniego urządzenia na devices.sensor.community jest Sensor ID dla miernika pyłu i on jest potrzebny do wykresów.

Dokładnie tak byłoby prościej -  próbowałem wywołać to api które powinno pobrać adres miernika, ale okazuje się że nie działa  tam teoretycznie powinny pojawić się wszystkie mierniki z prostokąta wskazanego koordynatami, ale jakoś tak to dziwnie działa że oba punkty są tożsame i zwraca null. Udało mi się pobrać dopiero jak zmniejszyłem precyzję zapytania do 3 miejsc po przecinku. Tak że lepiej to będzie ustawić na sztywno.

bfaliszek, 2022-01-05:
@bryan-pl Już jest w branch dev. Pliki .bin dla ESP32 i ESP8266. Dodałem tam w ustawieniach serwisów opcję wpisania ręcznie Sensor ID. Wyłączyłem też to automatyczne wyszukiwanie. Trzeba zawsze je podać ręcznie w ustawieniach. 
https://github.com/hackerspace-silesia/Smogomierz/tree/dev/firmware
prerelease_ESPxxxx_Smogomierz_2.8.2_PMS_build_5.01.2022.bin

bryan-pl, 2022-01-05:
@bfaliszek - Wrzuciłem, ustawiłem ID i wykresy się wyświetlają poprawnie. Ta opcja jest dużo prostsza niż wydłubywanie ID z api luftdaten przy starcie. DzięKi!

bryan-pl, 2022-01-05:
Zostaje jeszcze MQTT - sprawdziłem zarówno w ESP32 jak i w ESP8266 nie zapisuje ustawień serwera i portu.

bfaliszek, 2022-01-05:
OK, czyli zostało poprawienie MQTT, sprawdzenie czy działa influxDB oraz aktualizacja biblioteki HomeKi(tylko na ESP32)

bryan-pl, 2022-01-05:
Dzięki -czy mógłbym prosić o listę bibliotek aktualnie wykorzystywanych przez smogomierz do ESP32. Mam u siebie niezły bałagan sporo mi się wywala w trakcie kompilacji.

monteros007, 2022-01-10:
Wrzuciłem ten .bin prerelease do siebie na wemos d1:
1. wgranie ok
2. odpala się sieć wifi, konfiguracja okej
3. wejście w gui, zmiana ustawień na język polish oraz zmiana wysokości n.m.p
4. save
5. wysypanie się aplikacji
6. znów widoczna orig sieć smogomierza.

monteros007, 2022-02-18:
Nic się nie zmieniło, rozumiem, że nie będzie temat ogarnięty?

bfaliszek, 2022-02-18:
@monteros007 jaka konfiguracja(jakie ESP, czujnik pyłu i temp) i jaka wersja oprogramowania? Przy zmianie których ustawień pojawia się problem?

monteros007, 2022-02-18:
Wszystko jest w moim pierwszym poście, który zapoczątkowałem tutaj. Nic od tamtej pory w moim uczuciu się nie poprawiło. Przecież uczestniczyłem w konwersacji, trzeba tylko sobie poczytać. Używam Wemos D1 nie jest do niczego podpięty, po prostu po pierwszym zapięciu do wifi w ustawieniach zmieniach tylko na BME i PMS biorę ZAPISZ i wszystko się wysypuje. Jak wół stoi wyżej.

monteros007, 2022-02-18:
@bfaliszek okej tego nie testowałem, ostatnio prerelease jakiś. Wrzucam i dam zaraz znać. Odnośnie wemosa, testowałem na 3 różnych płytkach.

bfaliszek, 2022-02-19:
Podłącze tego Wemosa do komputera i w ArtuinoIDE włącz monitor portu szeregowego. Co tam wyświetla przy wysypaniu się?

monteros007, 2022-02-24:
Dobra, wgrałem https://github.com/hackerspace-silesia/Smogomierz/blob/dev/firmware/ESP8266/prerelease_ESP8266_Smogomierz_2.8.2_PMS_build_1.02.2022.bin na nodemcu, mam caly czas taki sam błąd i takie same objawy. Wysypuje się i wraca do ustawień fabrycznych. Pierwsza konfiguracja ok. Zaznaczam język polski, ustawiam BME280 i PMS, SAVE. Wywala.

bfaliszek, 2022-02-28:
URL wygląda ok, ale co pokazuje  monitor portu szeregowego z Arduino IDE? Tam powinno być info dlaczego urzązenie się zrestartowało i info co robiło chwilę wcześniej

monteros007, 2022-02-28:
W porządku ale jak z krzaków ͬ⸮⸮⸮⸮<⸮X zrobić użytek?

monteros007, 2022-02-28:
Proszę, logi w załączniku. Nic nie zdążyłem zrobić. 1 podłączenie do wifi, ou yeah i za chwilę restart do ustawień fabrycznych
[logi.txt](https://github.com/hackerspace-silesia/Smogomierz/files/8154636/logi.txt)
.

bfaliszek, 2022-03-01:
OK, ale to nie jest związane z zapisem, tylko masz restart zaraz po ustawieniu połączenia wifi. Nigdzie nie masz zmiany ustawień

bfaliszek, 2022-03-02:
Masz podłączone DHT22?

mati75, 2022-05-15:
Mam dokładnie to samo, bez podłączonych żadnych czujników.

austinroolon, 2022-05-30:
U mnie po aktualizacji (8266, BMP280, SPS30 nie da się ustawić poprawnie portów SPS oraz niektórych danych ThingSpeak. Wersja 2.7.9 SPS30 działa najlepiej, ale nie generuje poprawnie grafik na www. Do Thinkspeak raporty wysyła poprawnie. Ustawiłem wstępnie dane ThinkSpeak na wersji 2.7.9 i podniosłem do najnowszej SPS30. Gubi wpisy. Ręcznie wprowadzone ponownie nie zapamiętuje, albo zastępuje innym (nieznanym mi, tym samym za każdym razem). Póki co niestety jedynie działa u mnie 2.7.9 SPS30 - nie generuje poprawnie www i nie usypia SPS30 między pomiarami przez co szybko drenuje akumulator. Nawet na Sleep=1 (to już omawialiśmy). Gdzie zapisywane są ustawienia ?

dudus2000, 2022-06-13:
Czesc.
Esp8266 - bme280 - pms7003 
Wszstko sie sypie po probie zapisu ustawien MQTT
Mam pytanie - czy moglibyscie wydac porawke w formie ostatnio dzialajacego softu.
Czy z takim spreparowanym softem z wyzszym nr pozwoli przeprowadzic zdalna aktualizacje??
Niestety nie mam dostepu do urzadzenia a korzystam tylko z mqtt

hanskiepl, 2022-09-03:
Cześć, 
coś sie posypało u mnie też... nie wiem dlaczego. testowałem inne ESP ale jest tak samo.
Nawet po restarcie urządzenie (np zanik zasilania) gubi ustawienia wifi i trzeba sie logować do jego sieci i przestawiać.
Da się to jakoś obejśc na stałe ? kompilowac samem i na sztywno podać dane wifi ?
Czy może ja coś źle robię ?

hanskiepl, 2022-09-08:
Przeszedłem całą procedure programowania nowej płytki ESP8266 i to samo. nie zapamętje SSIDa do Wifi i po restarcie tworzy swoją sieć czekając na konfigurację.
Co jest grane ? jak temu zapobiec ?

hanskiepl, 2022-09-08:
Ok juz znalazłem rozwiązanie. Trzeba pozmienić kody żródłowe ESPASyncWifiManagera
https://github.com/alanswx/ESPAsyncWiFiManager/issues/102

> OK, ale to nie jest związane z zapisem, tylko masz restart zaraz po ustawieniu połączenia wifi. Nigdzie nie masz zmiany ustawień

Czy mógłbyś skompilowac ostatnią wersję ze skorygowanym WifiManagerem ?

LeanderLBN, 2022-11-04:
Jeśli dobrze rozumiem, to jest to rozwiązanie tego #154 problemu?
Zatem, czy można się w jakimś przewidywalnym czasie spodziewać poprawki i nowej wersji dla tych, którzy tylko wgrywają, a nie kompilują? :)

lukaszog, 2022-11-12:
@up podbijam pytanie, mam ten sam problem. Plus po około 24h smogomierz odłącza się od sieci wtedy trzeba wyłączyć zasilanie podłączyć jeszcze raz i ręcznie wybrać sieć. Korzystam z softu z branchu dev, z mastera nic nie chciało działać :). Czy gdzieś są zapisywane logi?

hanskiepl, 2022-11-13:
[prerelease_ESP8266_Smogomierz_2.8.2_PMS_build_1.02.2022-WifiFix.bin.zip](https://github.com/hackerspace-silesia/Smogomierz/files/9998090/prerelease_ESP8266_Smogomierz_2.8.2_PMS_build_1.02.2022-WifiFix.bin.zip)
Tu macie fixa resetującego się wifi.

LeanderLBN, 2022-11-16:
> [prerelease_ESP8266_Smogomierz_2.8.2_PMS_build_1.02.2022-WifiFix.bin.zip](https://github.com/hackerspace-silesia/Smogomierz/files/9998090/prerelease_ESP8266_Smogomierz_2.8.2_PMS_build_1.02.2022-WifiFix.bin.zip) Tu macie fixa resetującego się wifi.

Super! Dziękuję, działa.

hanskiepl, 2022-11-16:
> > [prerelease_ESP8266_Smogomierz_2.8.2_PMS_build_1.02.2022-WifiFix.bin.zip](https://github.com/hackerspace-silesia/Smogomierz/files/9998090/prerelease_ESP8266_Smogomierz_2.8.2_PMS_build_1.02.2022-WifiFix.bin.zip) Tu macie fixa resetującego się wifi.
> 
> Super! Dziękuję, działa.

Proszę uprzejmie :-)
Niestety pozostałych błędów to nie naprawia takich jak mqtt czy tych dziwnych linii na stronie chyba ustawień.

FilipTP, 2022-11-17:
> Proszę uprzejmie :-) Niestety pozostałych błędów to nie naprawia takich jak mqtt czy tych dziwnych linii na stronie chyba ustawień.

Ja też dziękuję :) na problemy z wifi pomogło, ale po zalogowaniu się mogę zapisać tylko ustawienia w jednym menu (np. zmiana języka na Polski, współrzędne, czujniki itd.), ale jak próbuję potem zapisać ustawienia na które serwisy ma wysyłać otrzymane dane to się wysypuje i nie mogę się już zalogować do czujnika. Jedyne wyjście to wgranie na nowo softu gdzie znowu mogę jedynie w jednym menu i jeden raz zapisać ustawienia.. chyba przydałoby się to jakoś systemowo ogarnąć..

lukaszog, 2022-11-30:
@FilipTP wifi działa prawidłowo tak? 
U mnie po aktualizacji niestety problem nadal występuje, po około 20h rozłącza.