hackerspace-silesia/Smogomierz #152: Błąd podczas wgrywania oprogramowania ESP32

mickelk, 2021-12-28:
Cześć, mam problem z ESP32, chce wgrać smogly i nie pokazuje mi się sieć wifi, po rozmowach z @bfaliszek kazał mi sprawdzić w monitorze portu szeregowego i mam taki efekt: 

`rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57`

Gdy wgram przykład z ArduinoIDE- działa, więc myśle że wgram smogly przez Arduino, przy pomocy tego poradnika, https://github.com/hackerspace-silesia/Smogomierz/blob/master/instrukcje/software.md#arduino---konfiguracja a po komplikacji dostaje info : 

> Smogomierz:134:10: fatal error: src/WiFiManager/ESPAsyncWiFiManager.h: No such file or directory
 #include "src/WiFiManager/ESPAsyncWiFiManager.h" // https://github.com/alanswx/ESPAsyncWiFiManager // 5.11.2021
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
exit status 1
src/WiFiManager/ESPAsyncWiFiManager.h: No such file or directory

>

bfaliszek, 2021-12-29:
Źle zflashowana pamięć w ESP32. Używałeś do wgrania pliku .bin esptool.py czy czegoś innego?
Na szybko w google znalazłem coś takiego 
"A:\> esptool.py.exe --port COM7 --baud 512000 write_flash --flash_size=detect 0x1000 .\Downloads\esp32-idf3-20210202-v1.14.bin" gdzie istotne jest 0x1000

mickelk, 2022-01-03:
Trochę czasu mi to zajęło, nie wiem czemu ale podczas wgrywania gotowego pliku bin się wysypuje. Próbowałem na 3 kompach i tylko smogomierz się wysypuje, gdy wgrywam HAA działa bez problemu. 
W każdym bądź razie udało mi się przez ArduinoIDE skompilować i wgrać - działa ale mam taki problem:

[https://ibb.co/tHFSV6k](url)

zamiast temperatury wyświetla się: {TEXT_HUMIDITY}:{Humidity} % 
próbowałem różnych ustawień ciągle ten sam efekt, przez zakładkę update wgrałem plik bin z wersji 2.7.1 temperatura się wyświetla prawidłowo, 
więc zaktualizowałem przez plik *.bin wersje z githuba Smogly_2.8.1_build_27.11.2021 i dalej ten sam problem.

bfaliszek, 2022-01-04:
Czyli jak rozumiem problem jest tylko ustawieniami MQTT? 
Brak pomiarów temp/humi pojawia się na ESP32 i poprawię to w kolejnej wersji.

mickelk, 2022-01-04:
ok, to czekam na update :) 
jak dodałem urządzenie do homekit to przy PM miałem nieznane mimo iż na www pokazywało, temp była 2C i jakaś dziwna wilgotność.

bryan-pl, 2022-01-04:
To jest ciekawe, bo mam właśnie ESP32 (devkitv1) i BME280 i działa bez problemu odczyt BME i PMS, natomiast są problemy w ogóle z wgraniem firmware na esp (tylko na jednym sofcie mi się udało, żeby zadziałało) i z zapisywaniem części ustawień w konfiguracji (m.in mqtt).

bfaliszek, 2022-01-04:
Udostępniłem prerelease_ESP32_Smogomierz_2.8.2_PMS_build_4.01.2022.bin - https://github.com/hackerspace-silesia/Smogomierz/tree/dev/firmware/ESP32
Poprawiłem tam obsługę BME280 i PMSx003. Teraz powinno się poprawnie dać ustawić piny na, których są podłączone i "{TEXT_HUMIDITY}:{Humidity}" nie powinno się już pojawiać. 
Przed wgraniem tego .bin zalecam kliknąć "Przywróć ustawienia domyślne"

mickelk, 2022-01-04:
Podeślesz Smogomierz.ino ?

bfaliszek, 2022-01-04:
@mickelk jest w branchu dev - https://github.com/hackerspace-silesia/Smogomierz/tree/dev tam aktualizuję wszystkie pliki przy wrzucaniu nowych testowych plików .bin

mickelk, 2022-01-04:
działa, ale w HomeKit nie pokazuje PM

[image]

bfaliszek, 2022-01-05:
@mickelk wiem, bo muszę trochę inaczej zrobić HomeKit. W tej bibliotece, która jest teraz używana nie ma obsługi poziomów PM. Jest już taka bardziej oficjalna i tam to działa, tylko muszę jeszcze przetestować zmienianie kodu, który jest przy parowaniu z urządzeniem Apple(żeby można było ustawić swój, a nie 111111111)

mickelk, 2022-01-06:
Jeszcze tak apropo HomeKit, przy temperaturze 0 lub mniej 
[image]

w HomeKit wyświetla się 

[image]

liske310, 2023-01-25:
Mam idealnie to samo... Arduino IDE jakiś przykład działa, a np. Twój program ma te same błędy co poprzednicy:
```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
...
```