hackerspace-silesia/Smogomierz #168: Część strony ustawień nie jest wyświetlana poprawnie

stsdc, 2022-11-24:
Po skompilowaniu i wgraniu softu na ESP8266 prawidłowo utworzył AP. Następnie po wprowadzeniu danych do WiFi Managera 
nastąpił Soft WDT reset, ESP znowu się uruchomiło w trybie AP. Po wprowadzeniu SSID i hasła udało się wejść na stronę ustawień, po zmianie ustawień i wciśnięciu SAVE znowu reset, jednak tym razem ESP pozostał w sieci i udało się znowu otworzyć stronę ustawień, jednak część strony nie renderuje się :cry: 

[image]
 
Co tu robić? :disappointed:

stsdc, 2022-11-26:
Wersja 2.7.9 działa poprawnie pod względem zapisywania ustawień.

radziejewicz, 2023-05-05:
Mam to samo issue, na binarce z brancha `dev` :)

bfaliszek, 2023-05-06:
@radziejewicz na pierwszy rzut oka to wygląda na źle podane piny dla miernika temperatury/wilgotności/ciśnienia w ustawieniach urządzenia. Przy okazji branch `dev` był w minionym tygodniu często aktualizowany, bo dodawałem sporo poprawek w kodzie. Ten kod, który jest dzisiaj(6.05) na `dev` jest już całkiem stabilny zarówno na ESP8266 jak i ESP32.

radziejewicz, 2023-05-06:
Niestety piny są poprawnie wpięte :)