hackerspace-silesia/Smogomierz #170: Brak wifi po flashu

radziejewicz, 2023-04-06:
Cześć na esp8826 wrzuciłem ESP8266_Smogomierz_2.8.1_PMS_build_27.11.2021.bin, lecz po flashu płytka nie wstaje, brakuje wifi :/ 

Flashuję tak:
```
$ esptool.py -p /dev/ttyUSB0 --baud 460800 write_flash --flash_size=detect 0x1000  ESP8266_Smogomierz_2.8.1_PMS_build_27.11.2021.bin esptool.py v4.5.1
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 30:83:98:85:a4:4e
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Flash will be erased from 0x00001000 to 0x000aafff...
Compressed 695568 bytes to 461673...
Wrote 695568 bytes (461673 compressed) at 0x00001000 in 11.7 seconds (effective 473.8 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

Ktoś podpowie co poszło nie tak?

lim3ra, 2023-04-28:
mam w sumie ten sam problem na ESP32

```
Chip is ESP32-D0WD-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: d4:d4:da:e5:49:20
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Erasing flash (this may take a while)...
Chip erase completed successfully in 12.7s
Compressed 1477968 bytes to 861887...
Wrote 1477968 bytes (861887 compressed) at 0x00000000 in 76.4 seconds (effective 154.8 kbit/s)...
Hash of data verified.

Leaving...
Staying in bootloader.

Firmware successfully flashed. Unplug/replug or reset device 
to switch back to normal boot mode.
```

bfaliszek, 2023-04-29:
@lim3ra hej! Sprawdź kod z brancha dev - https://github.com/hackerspace-silesia/Smogomierz/tree/dev
Właśnie do skompilowałem i wgrałem na ESP32 Dev Module i łączy się z WiFi
[image]

lim3ra, 2023-04-29:
hey @bfaliszek , dzieki za odpowiedz

Próbowałem sflashować 2.8.3 na moje ESP32 uzywając NodeMCU PyFlasher ale niestety nic to za bardzo nie daje. 

Spróbowałem też skompilować sobie brancha dev na swoim Arduino IDE ale non-stop rzuca się o jakieś dependency biblioteki np ESPAsyncWebServer.h nie wiem do końca jak sobie poradzić z ich brakiem (instalacja ESPAsyncWebSrv przez library manager'a nie pomaga)

masz jakiś pomysł czego mógłbym jeszcze spróbować?

bfaliszek, 2023-04-29:
Musisz ręcznie doinstalować brakujące biblioteki - https://github.com/hackerspace-silesia/Smogomierz/blob/master/instrukcje/software.md#arduino---konfiguracja - podpunkt 10

radziejewicz, 2023-04-29:
W razie czego brakujące libki (ja tak robiłem) pobrałem i wrzuciłem w katalog po prostu.

lim3ra, 2023-04-29:
Bardzo dziękuję za pomoc @bfaliszek, wszystko działa

bfaliszek, 2023-04-30:
@lim3ra czyli kod z brancha dev u Ciebie na ESP32? Czyli zostaje kwestia ESP8266. Problem w tym, że nie mam żadnego ESP8266, ale poszukam i postaram się to ogarnąć. 
Jestem też w połowie refactoringu kodu Smogomierza. Chcę to skończyć w przyszłym tygodniu i poprawić znane błędy.

lim3ra, 2023-04-30:
> @lim3ra czyli kod z brancha dev u Ciebie na ESP32? Czyli zostaje kwestia ESP8266. Problem w tym, że nie mam żadnego ESP8266, ale poszukam i postaram się to ogarnąć. 
> 
> Jestem też w połowie refactoringu kodu Smogomierza. Chcę to skończyć w przyszłym tygodniu i poprawić znane błędy. 

Hej, tak mogę potwierdzić, ze na ESP32 wszystko się kompiluje i wgrywa bez problemu. Działa tez wsparcie dla HomeKit.

radziejewicz, 2023-05-03:
O dzięki za fix, w sumie już też namierzyłem i poprawiłem w kodzie :) 

Co do refactoringu - to będziesz poprawiać błędy, czy dojdzie coś nowego?

lim3ra, 2023-05-03:
> Na razie przepisuję ten kod, który jest tak strasznie napisany, że aż wstyd mi na niego patrzeć. Większość już przepisałem na nowo. Chcę jeszcze poprawić błędy, które znajdę. Na razie nie planuję żadnych nowych funkcjonalności

Przy okazji - dzięki za Twoją pracę @bfaliszek 

świetny projekt, ja korzystam już od kilku lat

lipszczak, 2023-06-10:
Hej,
U mnie na ESP32 po flashu przez NodeMCU Pyflasher również nie pojawia się WiFi. 
W Arduino IDE jak próbuję samemu skompilować program to sypie błędami:
In file included from C:\Users\xxx\Documents\Smogomierz\Smogomierz.ino:556:
C:\Users\xxx\Documents\Smogomierz\src\smogly_asyncwebserver.h: In function 'String handle_root_processor(const String&)':
C:\Users\xxx\Documents\Smogomierz\src\smogly_asyncwebserver.h:94:3: error: 'takeTHPMeasurements' was not declared in this scope
   takeTHPMeasurements();
   ^~~~~~~~~~~~~~~~~~~
C:\Users\xxx\Documents\Smogomierz\src\smogly_asyncwebserver.h:94:3: note: suggested alternative: 'pmMeasurements'
   takeTHPMeasurements();
   ^~~~~~~~~~~~~~~~~~~
   pmMeasurements
In file included from C:\Users\xxx\Documents\Smogomierz\Smogomierz.ino:556:
C:\Users\xxx\Documents\Smogomierz\src\smogly_asyncwebserver.h: In function 'void handle_adv_mqtt_config(AsyncWebServerRequest*)':
C:\Users\xxx\Documents\Smogomierz\src\smogly_asyncwebserver.h:2044:5: error: 'takeTHPMeasurements' was not declared in this scope
     takeTHPMeasurements();
     ^~~~~~~~~~~~~~~~~~~
C:\Users\xxx\Documents\Smogomierz\src\smogly_asyncwebserver.h:2044:5: note: suggested alternative: 'pmMeasurements'
     takeTHPMeasurements();
     ^~~~~~~~~~~~~~~~~~~
     pmMeasurements
C:\Users\xxx\Documents\Smogomierz\src\smogly_asyncwebserver.h: In function 'void handle_api(AsyncWebServerRequest*)':
C:\Users\xxx\Documents\Smogomierz\src\smogly_asyncwebserver.h:3267:5: error: 'takeTHPMeasurements' was not declared in this scope
     takeTHPMeasurements();
     ^~~~~~~~~~~~~~~~~~~
C:\Users\xxx\Documents\Smogomierz\src\smogly_asyncwebserver.h:3267:5: note: suggested alternative: 'pmMeasurements'
     takeTHPMeasurements();
     ^~~~~~~~~~~~~~~~~~~
     pmMeasurements
C:\Users\xxx\Documents\Smogomierz\Smogomierz.ino: At global scope:
C:\Users\xxx\Documents\Smogomierz\Smogomierz.ino:1078:3: error: expected unqualified-id before 'if'
   if (need_update == true && deviceSettings.autoUpdate) {
   ^~
C:\Users\xxx\Documents\Smogomierz\Smogomierz.ino:1086:10: error: expected constructor, destructor, or type conversion before ';' token
   yield();
          ^
C:\Users\xxx\Documents\Smogomierz\Smogomierz.ino:1088:19: error: expected constructor, destructor, or type conversion before ';' token
   pm_calibration();
                   ^
C:\Users\xxx\Documents\Smogomierz\Smogomierz.ino:1091:3: error: expected unqualified-id before 'if'
   if (!strcmp(sensorsSettings.dustModel, "PMS7003")) {
   ^~
C:\Users\xxx\Documents\Smogomierz\Smogomierz.ino:1112:3: error: expected unqualified-id before 'if'
   if (strcmp(sensorsSettings.dustModel, "Non")) {
   ^~
C:\Users\xxx\Documents\Smogomierz\Smogomierz.ino:1169:5: error: expected unqualified-id before 'else'
   } else {
     ^~~~
C:\Users\xxx\Documents\Smogomierz\Smogomierz.ino:1205:3: error: expected unqualified-id before 'if'
   if (airMonitorSettings.enabled) {
   ^~
C:\Users\xxx\Documents\Smogomierz\Smogomierz.ino:1215:3: error: expected unqualified-id before 'if'
   if (luftdatenSettings.enabled or aqiEcoSettings.enabled or smoglistSettings.enabled) {
   ^~
C:\Users\xxx\Documents\Smogomierz\Smogomierz.ino:1224:3: error: expected unqualified-id before 'if'
   if (thingSpeakSettings.enabled or influxDBSettings.enabled or mqttSettings.enabled) {
   ^~
C:\Users\xxx\Documents\Smogomierz\Smogomierz.ino:1235:3: error: expected unqualified-id before 'if'
   if (current_REBOOT_Millis - intervals.previousRebootMillis >= intervals.reboot) {
   ^~
C:\Users\xxx\Documents\Smogomierz\Smogomierz.ino:1263:1: error: expected declaration before '}' token
 }  // loop() - END
 ^

exit status 1

Compilation error: 'takeTHPMeasurements' was not declared in this scope

lipszczak, 2023-06-10:
Odpowiem sam do siebie. 
U mnie flashowanie przez NodeMCU Pyflasher z pliku bin nie działało. Ten sam plik bin, wgrany przez ESPHomeflasher, na tę samą płytkę działa. Nie jestem specem, ale czytałem że NodeMCU nie wgrywa niezbędnych danych bootloadera, więc wgrywając z pliku bin, musimy mieć w pliku binarnym te brakującą część. ESPHomeflasher wgrywa tę część dump (czy jakoś tak) z automatu.
Teraz walczę z ustawieniami, żeby wyświetlało wyniki pomiarów, bo mój smogomierz póki co pokazuje same 0