hackerspace-silesia/Smogomierz #171: ESP32 flashed devices dont work

andygodber, 2023-05-08:
Apologies for writing in English.
Im unable to use the ESP32 image on any devices.
I have used the NodeMCU-pyFlasher and a selection of different ESP32 .bin files (inc pre-release 2.8.3).
The files appear to be written by the flash tool, as the % indicator increases and says completed successfully; the tool reports the connected ESP as ESP32-D0WDQ6-V3.
If I look at the serial output in Arduino IDE, it seems to be in a boot-loop:

invalid header: 0x55544954
inva�ets Jul 29 2019 12:21:46

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
invalid header: 0x55544954
invalid header: 0x55544954
invalid header: 0x55544954
invalid header: 0x55544954
invalid header: 0x55544954
invalid header: 0x55544954
invalid header: 0x55544954
invaets Jul 29 2019 12:21:46

(I have successfully flashed and used the 8266 version.)

andygodber, 2023-05-08:
Confirmed that I tied with the pre-release.

[image]

Im also having trouble trying to complile from the code (missing some libraries - I added BME280 from Ada (as well as BMP) but still showing an error with bmeStatus as not being defined).

lipszczak, 2023-06-11:
Try to flash .bin with ESPHomeflasher. I had the same issue with NodeMCU flasher.